dattas/OctoPrint-DetailedProgress #2: Error:No Checksum with line number, Last Line: 29

oxcxsxixbxaxn, 2016-10-05:
Printing from Cura 2.3 via Octoprint plugin and all was well until I installed this Detailed-Progress plugin
Now the gcode is sent over and the bed and hotend heat, all looks well until it gets up to temperature but instead of printing goes into an endless loop, somehow tripping over the M117 commands.
...
Send: N30 M117 ETL 01:01:17*76
Recv: Error:No Checksum with line number, Last Line: 29
Recv: Resend: 30
Recv: ok
Send: N30 M117 ETL 01:01:17*76
Recv: Error:No Checksum with line number, Last Line: 29
Recv: Resend: 30
Recv: ok
Send: N30 M117 ETL 01:01:17*76
Recv: Error:No Checksum with line number, Last Line: 29
Recv: Resend: 30
Recv: ok
...
The Marlin firmware on the ramps 1.4 usually supports M117 without problems.

The DetailedProgress plugin was installed as normal through the octoprint gui
Any ideas?

dattas, 2016-10-05:
Are you using RepetierFirmware? It doesn't like certain characters like : or %, you'll want to read the readme on how to change it. If not, please provide details on your printer.

oxcxsxixbxaxn, 2016-10-05:
Delta Printer with Ramps 1.4 board running Marlin firmware, LCD is REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

dattas, 2016-10-05:
What version of marlin are you running? I have the same board (running in not delta) with the same screen.

oxcxsxixbxaxn, 2016-10-05:
Its 1.0.0, modified with some additional calibration routines

oxcxsxixbxaxn, 2016-10-05:
I think it's the colon ":" in the command.  
Where does Plugin Manager put the config for this plugin?  I guess I could modify this to 
{hours:02d}h{minutes:02d}m{seconds:02d}s

dattas, 2016-10-06:
Try version 0.1.3 (  commit 6aece654f889941be9a4c35bd1929c06ab7c76c3 ) I've removed all special characters which should help firmware compatibility

krwynn, 2016-10-09:
Just as an FYI. I use Repetier v1.6.2, Slice with Slic3r 1.2.9, Marlin 1.1.0 RC7 using a REPRAP Discount Full Graphic Smart Controller and had no issues with the previous version of the Detailed Progress plugin.

tonytemo, 2018-05-16:
Hello guys :) I'm making a CNC machine( 2D printing Only) as a graduation Project ....I'm using Ramps ,  ARDUINO MEGA , Marlin Software V1.1 and Repetier program ...when I upload a G-Code File the Repetier program starts but at a certain point the machine stops  and gives me this error .

Error:No Checksum with line number, Last Line: 29
Recv: Resend: 30

What should I do to avoid it ????(i use inkscape and Makercam to upload the design to Repetier )
thanks :)

closed 2024-10-20 by dattas