dattas/OctoPrint-DetailedProgress #5: Error when changing eta_strftime

calonmerc, 2016-12-01:
I changed the eta_strftime to:
`eta_strftime: %H:%M:%S`

And now get this error on octoprint runtime: 
`found character '%' that cannot start any token
  in "/home/pi/.octoprint/config.yaml", line 107, column 19`

Line 107 in my config.yaml is the eta_strfime command.

dattas, 2016-12-01:
@calonmerc for yaml, you must escape %, try this:

calonmerc, 2016-12-01:
It's not escaped in the default config though.... Gina just mentioned escaping the : characters.

calonmerc, 2016-12-01:
And aren't the %H, %M, and %S characters to be replaced by the plugin? So wouldn't escaping them break that?

calonmerc, 2016-12-01:
I tried your solution and got the same error. Putting the entire thing in quotes, I now get an error about the line after which is default and I have not changed.

calonmerc, 2016-12-01:
In summary... 

I've tried the following all resulting in an error still:
`eta_strftime: %H:%M:%S`
`eta_strftime: %%H:%%M:%%S`
`eta_strftime: %H%M%S`
`eta_strftime: "%H:%M:%S"` this one resulted in the error being on the completion line of:
`    messages:
      - {completion:.2f}% complete`

calonmerc, 2016-12-01:
Changing the entire thing to:
` detailedprogress:
    time_to_change: 5
    eta_strftime: "%H:%M:%S"
    etl_format: "{hours:02d}:{minutes:02d}:{seconds:02d}"
    messages:
      - "{completion:.2f}% complete"
      - "ETL: {printTimeLeft}"
      - "ETA: {ETA}"`

Solved the issue.... so I HIGHLY suggest changing it to this in the default config. Even pasting the default config back in still didn't fix it.

foosel, 2016-12-02:
See also the [YAML primer](http://docs.octoprint.org/en/master/configuration/yaml.html#scalars):

> For most scalars you don’t need any quotes at all, but if you need to define some piece of data which contains characters that could be mistaken with YAML syntax you need to quote it in either double `"` or single `'` quotes for the YAML file to stay valid. As simple rule of thumb, if your data contains any of these characters `:-{}[]!#|>&%@` better quote it.

1n5aN1aC, 2017-01-09:
Yup, this does seem to fix it!

Issue should probably be closed as well.

closed 2024-10-20 by dattas