dattas/OctoPrint-DetailedProgress #8: Output to LCD

azalanono, 2017-02-09:
I installed the DetailedProgress plugin and find it very helpful. 

My 3D printer uses Marlin Firmware and has a 20x4 LCD installed. Everything is fine, but I noticed that  following things need either explanation or correction:

- The LCD displays an ETA being far away from reality.  I started a printjob with ~45 minutes running time and see an ETA of some days(!)  This is absolutely wrong. When ETA calculation is a problem, it should be better removed.  It would be more helpful to see the IP address instead.

- The format of ETA is very unusual. I would recommend something like this: "DD.HH:MM:SS"

- ETL is displaying a correct value, but I would suggest the same format here, too.  "DD.HH:MM:SS"

- The detailed done percentage has a funny output:  "99.9p .....".  Why not use the "%" sign, it definitely works with M117 command? 

- Displaying the IP address periodically would be a nice additional output. At the moment, the IP is only shows when the printer is idle.

It would be the best to extend the plugin by a settings editor. 
This will allow to turn particular messages on/off as needed. In this editor it should also be possible to enter the format string to be used.

cosmith71, 2017-02-17:
I agree with this.  I'm a noobie...is there a way to configure the way the messages are displayed?

Thanks,

--Colin

cosmith71, 2017-02-17:
OK, somehow got the configuration to work.  I put the sample config on the front page into config.yaml.  Something then modified it and took out all the comments and set the cycle time to NULL.  I was able to modify the new, mystery configuration to my liking.  I'm not really sure what happened, but it's working now.  I have a % instead of the 'p'.  I think there may be something flaky with the way it sets up its configuration.

jasoncross, 2017-04-10:
I'm trying to do the same, but can't find where I update the configuration.  Can you point me in the right direction?

mkotek, 2017-05-03:
I have the same issue as cosmith71, also saving the defaults has caused the comments to be removed, but the configuration using % as suggested in readme does not work in my case - I am getting errors (as seen in terminal). I am trying to see, if the amended formatting written in the init file directly will be showing up correctly.
Any ideas, why the standard config works from init file, but the same entries as defaults in config.yaml do not?

dipusone, 2017-09-09:
Thanks @mkotek this settings saved me!

christhesmith, 2018-01-16:
@mkotek In playing around with this plugin over the last few days, I discovered that for my Maker Select v2.1 running Repetier, I was able to get the percent sign to display by using three '%' symbols in the config like so:

'{completion:.2f}%%% complete'

That said, with this printer/firmware, it appears that the colon ':' cannot be displayed in the same way (perhaps interpreted as a comment, like the semicolon?), so I changed the formats for the other two messages to what's below, and they seem to work properly now:

closed 2024-10-20 by dattas