dattas/OctoPrint-DetailedProgress #11: Wrong ETA

LundSoftwares, 2017-05-10:
Got an issue with the ETA that´s sent to the display.
I live in Sweden and we´re GMT +2 in the summer, but the time displayed is always in GMT, that means 2hrs short. If real ETA is 11.00 local time, display shows 9.00.
If I run "date" in the SSH shell on my Ocopti Raspberry Pi my correct localtime is displayed so I assume thoose settings are correct.
Any ideas?

Givemeyourgits, 2017-05-30:
My ETA is strange as well, just see stuff like ETA 18 03 54 Day which I have no idea what it means.  ETL seems to be updating properly.

Boby71, 2017-06-02:
Folks, I had the same problem with my OctoPi installation. I just changed the time-zone on my Raspberry and now the ETA is fine. You can do this via SSH/command line using the command:
"sudo rapi-config", then choose the internalization options and set the correct time zone. There's even a gui guiding you through the process helping you select the right city. Not even a reboot is necessary.

Cheers, Boby

P.S.: I guess this only works on a Raspberry Pi...

Boby71, 2017-06-02:
@Givemeyourgits, ETA means "estimated time of arrival" = the day and the time when your piece should be finished. Example: My currently running print says: ETA 18 04 15 Day 02. This means, that it will be finished on the day 02 (2nd June), 18:04:15 (or 6:04:15 pm)

LundSoftwares, 2017-06-04:
@Boby71 
The thing is that I´we already changed my setting to correct TimeZone with Raspi-config, 
This is my Output when doing it again:
`pi@octopi:~ $ sudo raspi-config

Current default time zone: 'Europe/Stockholm'
Local time is now:      Sun Jun  4 17:28:14 CEST 2017.
Universal Time is now:  Sun Jun  4 15:28:14 UTC 2017.
`
But my ETA in the display still shows my time in UTC and not in CEST that is set.

justinotherguy, 2018-03-09:
G'day!

I've had the same issue; after setting the timezone as suggested by @Boby71  my issue is solved :)

Thanks - have a good one!

LeightonSolo, 2024-10-20:
Having the same issue, although I'm using Octo4a so not sure how to set the correct time zone. The time zone on the android phone is correct and doesn't seem to affect the timezone assumed by detailed progress. Any ideas?

closed 2024-10-20 by dattas