dattas/OctoPrint-DetailedProgress #14: Print sequence stalls during preheat if custom display settings are set.

bagobones, 2017-07-26:
I cut and past the current example config into my config.yaml plugins section and regardless of what is in it my the printing process hangs at pre-heat. In the log there are messages about no ETL yet or something along those lines.

If the section is removed completely everything works fine and the default ETA values are displayed.

All I really want to change is the ETA format or even to remove that message.

I have also noted that if I cut and past the example as is and reboot octoprint alters the config..The double quotes turn to single quotes and the "time_to_change: 10" line gets stripped altogether.

deddolo, 2017-09-05:
Same problem here. The terminal complains that there is no checksum in the ETL line and to resend. The completion message which precedes this one works fine.

bagobones, 2019-01-06:
The example configuration contains the ":" character which the plugin appears to not deal with correctly when using the custom settings. 

If you remove it BOTH from the formatting strings and from the messages it works.

closed 2024-10-20 by dattas