dattas/OctoPrint-DetailedProgress #22: hello,

bionno, 2018-01-21:


bionno, 2018-01-21:
Hello,
like in the title, where is the folder/files to modify the OUTPUT ?

Thanks,
A.

sfeakes, 2018-01-21:
Looks like it goes in the plugin section of octoprint's config.yaml

So copy the settings listed on the main page, (except the _plugins:_) statement, and paste them in Octoprint's config.yaml under the _plugins:_ section. Then modify as you want.

Octoprints's config.yaml is usually in /home/<user_name>/.octoprint/config.yaml

I can't test this as I'm printing at the moment.

sfeakes, 2018-01-21:
I just tried this.  The config.yaml file is the place for the configuration, but be careful with the settings listed on the main page, as it looks like they are incorrect.

Using the settings I get checksum error when it tried to print.

```
Send: N11 M117 ETL: 01:59:08*118
Recv: Error:No Checksum with line number, Last Line: 10
```

Without them
```
Send: N11 M117 ETL 01h59m08s*58
```

So it looks like my printer control board doesn't like the `:` character, which would make sense since it's a special character in GCODE.
So you'll probably need to play with the `eta_strftime` and `etl_format ` format strings.

closed 2024-10-20 by dattas