dattas/OctoPrint-DetailedProgress #39: CR-10S Pro support

santanig5, 2019-03-24:
Greetings,
I have just upgraded to the CR-10S Pro, the status display does not appear to support your plugin, am I doing something wrong or would this be a future enhancement?
Really love the info when printing.

Cheers!

rgviewer, 2020-04-11:
I was wondering if you ever figured it out for the CR10S-Pro. I also have the Pro model and can't get it to display ETA or anything when using OctoPrint.

santanig5, 2020-04-12:
To some extent, my display shows the progress % and time printed, but no ETA.
I have these two plugins active, not sure which one did the trick or if I had to change anything in my gcode for it to work this way.
If you find something better, please let me know.
https://github.com/cesarvandevelde/OctoPrint-M73Progress
https://github.com/celogeek/OctoPrint-ProgressBasedOnTime

closed 2024-10-20 by dattas