dattas/OctoPrint-DetailedProgress #40: New maintainer?

stantond, 2019-08-20:
There are a lot of great improvements waiting to be merged, and other branches without PRs because the existing ones have been waiting a long time.

@dattas if you no longer have the time or interest to maintain the project, could you ask one of the other active contributors if they'd like to take this over, so the OctoPrint community can still use these great features without missing bits and pieces of new functionality across other branches?

hapklaar, 2019-12-08:
I found this fork to be the most active/recent

https://github.com/eiannone/OctoPrint-DetailedProgress

stantond, 2019-12-08:
I've seen another one that had a proper settings page for the config, and some other features dotted around. Problem is the OctoPrint plugin manager points here, where PRs have been ignored for a few years, so there's no single, central place to merge all improvements, which is a shame - lots of effort going unrecognised and under-utilised.

hapklaar, 2019-12-08:
I know, it's a shame. Hope someone can take this up as the plugin itself is a great addition to Octoprint.

tpmullan, 2020-04-28:
I am willing to take up development of this plugin.  I use it with every print I do.
I have merged many updates from others and have the changes available here: https://github.com/tpmullan/OctoPrint-DetailedProgress

stantond, 2020-05-02:
@tpmullan that's exciting, would be great to see this move forwards. Any plans to go through the old PRs and Issues to point towards your fork and add implement any suggestions/fixes/PRs to your fork?

tpmullan, 2020-05-02:
@stantond I have merged many of the pull requests that I have seen here.  Is there anything in particular that I am missing and can add?

stantond, 2020-05-02:
None in particular, just in case someone's raised something that's still relevant, and to redirect people if they come here via search engine

stantond, 2020-05-02:
I'll close this issue, seems it's solved!

closed 2020-05-02 by stantond